martinroger/VXDash #13: Find a better way to manage custom parts

martinroger, 2023-04-20:
Currently custom parts and footprints mostly managed through... well they are not managed. Attempts were made to add a submodule but this is a bit too convoluted to maintain new updates. Add a project sub-folder and point Samacsys' Kicad Library app to it.

Any suggestions welcome.

Alex-XNR, 2023-04-20:
I've typically worked with custom libraries in a separate repository. Each user would have their own working branch but would regularly update from/merge into main while checking for conflicts (i.e. not removing or reversing other users work). 

Any parts that are added, or any included KiCad parts that need modifying, are copied into this library so that everyone has the same library parts. Also, when KiCad is updated, any included library modifications you've made will likely be removed...

Folder structure something like this:
KiCad Library
   ->3D Models
   ->Symbols
   ->Footprints

martinroger, 2023-04-20:
> I've typically worked with custom libraries in a separate repository. Each user would have their own working branch but would regularly update from/merge into main while checking for conflicts (i.e. [...]

How would you refer the library repository to the main project library in Kicad ? Just configuring it as an additional library in Kicad (as opposed to a Project library) ?
I could/should set that up for sure.

Alex-XNR, 2023-04-21:
Yes exactly right, they would get added as global libraries, then for any future projects you're set up ready to go ;). For example, I have a custom symbol library for passives, transistors, ICs, etc,. But that's only my personal preference

martinroger, 2023-04-21:
> Yes exactly right, they would get added as global libraries, then for any future projects you're set up ready to go ;). For example, I have a custom symbol library for passives, transistors, ICs, etc,. But that's only my personal preference

Right, I'll try and set up that next then !

closed 2023-04-24 by martinroger