martinroger/VXDash #22: Improve ground thermal relief and stitching

Alex-XNR, 2023-04-25:
It's advisable to have thermal relief on any pad that connects to a large ground plane. It might cause difficulty when soldering if they are a solid connection.

Include ground vias at regular intervals around the ground plane and at any signal later changes to keep the ground potential as uniform as possible on the PCB.

martinroger, 2023-04-25:
I usually make sure that the PTH pads have thermal reliefs, is that also something that should be done for SMD pads ?

Alex-XNR, 2023-04-25:
I would recommend it. If hand soldering parts you'll need to put in more heat if using solid connections, which will risk damaging components.

martinroger, 2023-04-25:
I was actually thinking of breaking the bank and getting one of those chinese reflow oven or a 200mmx200mm hot plate. Arguably I put some really tiny SMD components (like the chips for the power supply stages). Would thermal reliefs help at all in those cases ?
Just a bit worried on the compromise of good grounding vs thermal reliefs on most large SMD ground pads.