martinroger/VXDash #23: Implement constant current source for Resistive sensor

martinroger, 2023-04-25:
- For sensors in the usual range (0-250Ohm) : max 13.2mA
- For sensors in the mid range (0-500Ohm) : max 6.6mA (use high-K voltage divider and allow 0-5V instead ?)
- For sensors in the high range (0-2K5) : allow 0-5V but imperatively use a 5V -> 3.3V divider with high values for the divider ? Still 2mA max

- Look into high-side current sources possibly based on Op-Amps ?
- Found some stuff on low A LED drivers like NCR401U, worth it or can still be bettered ?

martinroger, 2023-04-30:
Alright, so progressing this a little bit with commit 0abb7f7 : 
- I added a hierarchical sheet in the schematics using the LM358BA dual op amp (because it already has Rf and Ri integrated)
- There is a reference 4.096V fed to the op-amp by a LM4040-4.1 "Zener"
- Rsense is set in this implementation to 220Ohm to give a theoretical supply of 4.1mA, which accounting for most of the voltage drops here and there means that the maximum readable sensor resistance Rload of approximately 800Ohm. Practically it may be more but it starts to be really out there in various saturation zones

A concern remains on the quality of the 5V supply, as the "constant" current is almost more dependant on fluctuations on the 5V than anything else.
Also the protection Schottky has to be rethought, because its Vf will remain an unknown (even if at those low amps it might be relatively stable, even against temperature variations ). A 3.3V Zener  to GND instead ?

[image]

[image]

Alex-XNR, 2023-05-23:
Minor changes to the OpAmp design:
- Reference divider added to help with potential 5V supply variation.
- Current set voltage reduced to 100mV to increase measurement headroom.
- MOSFETs used instead of BJT so the base current doesn't influence the constant current.
- Iset resistors moved outside of the hierarchical sheet.

[image]

closed 2023-05-23 by Alex-XNR