martinroger/VXDash #31: Constant current LM358 sources are feeding too much current through Rset

martinroger, 2024-04-29:
I'm picking the project back up now that work is easing up a bit, on the Martin-dev branch.

When placing a 270Ohm load on one of the four AnalogR constant source circuits, I observe more current than expected through the load. I checked the circuitry and components position, all seems OK, however I see a larger than expected deltaV between IN1+ and IN1- . If I open-circuit, or nearly so, the Rload, the dV returns to about 100mV as expected.

I'm double-triple-checking the components I bought, soldering and everything, and sharpening up on LTSpice to see if the issue is not somewhere in my understanding of OP Amps and their limits, but @Alex-XNR if you have some insights it would be interesting to me. 

Essentially with Rset at 22Ohm today it doesn't really read right above 250Ohm.
[Calibration reference.xlsx](https://github.com/martinroger/VXDash/files/15155680/Calibration.reference.xlsx)
Attached are some measurements. It is fairly uniform across all the 5 similar sensing channels, I haven't yet investigated the one with 51R Rset for higher impedance loads, but I expect a similar behaviour.

martinroger, 2024-04-30:
Tasks : 

- [ ] Order AD8542 in MSOP and swap on the benchboard
- [ ] Update Schematic & PCB
- [ ] Change the VGPIO divider for something much larger. At higher Rsense there is a significant amount of current leaking through there (so 51k-27k instead of 5k1 2k7)
- [ ] Redo measurements and implement in code the compensation table

Alex-XNR, 2024-05-01:
Ahh, you are correct about the opamp common mode. 
First thing you could try is bring the opamp into CM range. Set Vref on the non-inverting to 2.5V, recalc for the Rset (try 510R at 2.5V ~5mA). Should still be some headroom for upto ~400ohm.
For round two, like you suggest, use a rail-to-rail opamp with full suppy CM range.

martinroger, 2024-05-02:
Yeah, what I will do is probably a resoldering exercise on the bench board directly for the OpAmp, turns out I have gotten quite alright with the hot air gun thing ! In my later commits I added a bunch of LTSpice simulations with the various replacement Op-Amps, I guess in the end I will keep the 5V supply and 4.9V Vref, bulk up the voltage divider from 5V ->3.3V on the GPIO side to minimize stray currents going to ground and run like this. Definitely also looked at setting a larger dV and Rset pair to get the same current but further away from CM, unfortunately I don't have those as 0603 package available to solder right now !

martinroger, 2024-11-04:
Closed by #35

closed 2024-11-04 by martinroger