martinroger/VXDash #33: Possibly inverted P-MOSFET on Power switch stage ?

martinroger, 2024-10-05:
Hey @Alex-XNR , I was checking some things after dusting off the project a bit, and I noticed - just now - that actually the benchboard starts itself regardless of the ignition signal value. I dug up the schematic, and I think I found what is wrong but I might need to double check with you : 

[image]

On there, shouldn't Q2 have D and S reversed ? And R65 and D10 should connect to Power_Fused on the left side ? When I simulate it, it seems to work but I thought I would double check with you before correcting that !

martinroger, 2024-10-06:
Candidate correction in 9ae1b542eb9109c802848487aa74be222306fe4f

Alex-XNR, 2024-10-10:
Hi @martinroger, yes looks like an error 🫣 As you've said, flip drain and source, keep D10 + R65 between G-S.

martinroger, 2024-10-10:
@Alex-XNR Imagine my face when I finally wondered why it was starting without me setting the ignition pin on...

[image]

Some like this looks OK to you ?

martinroger, 2024-11-04:
Closed by #35

closed 2024-11-04 by martinroger